tiagocoutinho/modbus-proxy #2: Connecting to SolarEdge Inverter unstable

magpern, 2021-01-24:
* ModBus TCP proxy version: 0.2.0
* Python version: The Docker one
* Operating System: Derbian

### Description

I have a SolarEdge inverter which only allows one connection at a time.
modbus-proxy would be the bridge between the SolarEdge inverter and a battery storage system and HomeAssistant as monitoring platform.

The SolarEdge inverter seams to have a somewhat unstable Modbus TCP implementation. If the client does something wrong or not follow the undocumented connection routines, it will stop responding. You will loose connection and you will not regain it without total restart on modbus-proxy.

So in other words. Modbus-proxy does connect to SolarEdge, it proxys data for some time, a few hours up to a day. Then it stops.
This is all you see in the trace file.

`2021-01-24 20:41:57,191:   INFO:modbus-proxy.Client(192.168.1.87:35512):new connection,
2021-01-24 20:41:57,191:  ERROR:modbus-proxy.Client(192.168.1.87:35512):ConnectionResetError(104, 'Connection reset by peer'),
2021-01-24 20:43:02,201:   INFO:modbus-proxy.Client(192.168.1.87:35518):new connection,
2021-01-24 20:43:02,202:  ERROR:modbus-proxy.Client(192.168.1.87:35518):ConnectionResetError(104, 'Connection reset by peer'),
2021-01-24 20:44:07,207:   INFO:modbus-proxy.Client(192.168.1.87:35524):new connection,
2021-01-24 20:44:07,207:  ERROR:modbus-proxy.Client(192.168.1.87:35524):ConnectionResetError(104, 'Connection reset by peer'),
2021-01-24 20:45:12,223:   INFO:modbus-proxy.Client(192.168.1.87:35530):new connection,
2021-01-24 20:45:12,223:  ERROR:modbus-proxy.Client(192.168.1.87:35530):ConnectionResetError(104, 'Connection reset by peer'),`

After a reboot of the docker instance, it connects again

`2021-01-24 20:45:12,223:  ERROR:modbus-proxy.Client(192.168.1.87:35530):ConnectionResetError(104, 'Connection reset by peer'),
2021-01-24 20:45:52,755:   INFO:modbus-proxy:Starting...,
2021-01-24 20:45:52,760:   INFO:modbus-proxy:Ready!,
2021-01-24 20:46:17,242:   INFO:modbus-proxy.Client(192.168.1.87:35536):new connection,
2021-01-24 20:46:17,242:   INFO:modbus-proxy.modbus(192.168.1.8:1502):connecting to modbus at ('192.168.1.8', 1502)...,
2021-01-24 20:46:17,245:   INFO:modbus-proxy.modbus(192.168.1.8:1502):connected!,
2021-01-24 20:46:17,507:   INFO:modbus-proxy.Client(192.168.1.87:35538):new connection,
2021-01-24 20:46:17,507:   INFO:modbus-proxy.Client(192.168.1.87:35536):ConnectionClosedError('Client disconnected'),
2021-01-24 20:47:22,511:   INFO:modbus-proxy.Client(192.168.1.87:35544):new connection,
2021-01-24 20:47:22,672:   INFO:modbus-proxy.Client(192.168.1.87:35544):ConnectionClosedError('Client disconnected'),
2021-01-24 20:48:27,676:   INFO:modbus-proxy.Client(192.168.1.87:35550):new connection,
2021-01-24 20:48:27,837:   INFO:modbus-proxy.Client(192.168.1.87:35550):ConnectionClosedError('Client disconnected'),`

### Interesting observations
The company Victron has an operating system called Venus OS. They have a feature that connects to SolarEdge inverters and retrieves the data.
It has never really worked, until they found a problem in their code and fixed it (2021 jan 21 beta release). After this fix, Venus OS and SolarEdge never drops the connection, before this fix, it dropped constantly (max an hour)

It is discussed here https://github.com/victronenergy/venus/issues/760 maybe this could help.
The fix is here https://github.com/victronenergy/dbus-fronius/commit/6ecc1d01840d327ed27607d03ce1be070c3f6836 although their code is C++, but the problem seems to be the same.

tiagocoutinho, 2021-01-25:
Thanks for the detailed report and for your patience.

It looks like the bug in Venus OS was related to Qt. They were calling the Qt `disconnect()` which as disconnecting the socket object from all qt signals instead of calling `disconnectFromHost()` which actually closed the socket... so my guess is it is not related.

I made some changes in master trying to take into account the following:
* Some Modbus slaves decide to close the TCP connection in some circumstances (ex: connection is idle for some time, connection is active for a maximum time, excessive request frequency, etc)
* Some clients close their end of the socket after each req/rep

Actions taken:
* when attempting to communicate with slave make a second attempt in case the socket was closed by the slave
* close the connection to the client if any error occurs
* handle clients which close sockets after each request
* make sure each message on both ends are full TCP messages (sometimes TCP packets can be broken into smaller pieces so we need to make sure we read a complete modbus message before sending it to the other side)
* add a modbus timeout
* better logging

I kindly ask that you test the latest master with an added parameter `--log-level=DEBUG`.
If the problem persists could you send back the log? Thanks in advance.

magpern, 2021-01-25:
I must say. You level of engagement and support... it is... as if you were me !!!

I have installed the latest version and is running it in a docker.
At first glance, it looks like it is working, we'll see if it survives the night. It has not in the past.

I had to alter the dockerfile to include the debug option

`CMD modbus-proxy -b "${MODBUS_BIND}" --modbus="${MODBUS_ADDRESS}" --log-level=${LOG_LEVEL}`

and in `docker-compose.yml` i do this:

```JSON
version: '3'
services:
  tcp_modbus_proxy:
    container_name: tcp_modbus_proxy
    image: mapern/tcp_modbus_proxy
    restart: unless-stopped
    build: modbus-proxy/.
    environment:
      - MODBUS_ADDRESS=tcp://192.168.1.8:1502
      - MODBUS_BIND=tcp://0:1502
      - LOG_LEVEL=DEBUG
    ports:
      - 502:1502
`

Is the trace file stored somewhere? As it is now, the trace file is just accessible through docker/portainer.
Can it be saved to disk? If yes, a data folder in the docker would be needed, because it is not possible to open a terminal in this docker image. (ENTRYPOINT missing?)
I don't know how many rows of trace are accessible through docker/portainer, but last time it stopped working I saw it too late, and the trace was not accessible.

So far the trace looks like this:

2021-01-25T22:54:22.747917000Z 2021-01-25 22:54:22,747:  DEBUG:modbus-proxy.Client(192.168.1.87:46806):received b'\x01\xd9\x00\x00\x00\x06~\x03\x9c\x85\x004',
2021-01-25T22:54:22.749463000Z 2021-01-25 22:54:22,747:  DEBUG:modbus-proxy.ModBus(192.168.1.8:1502):sending b'\x01\xd9\x00\x00\x00\x06~\x03\x9c\x85\x004',
2021-01-25T22:54:22.764356000Z 2021-01-25 22:54:22,763:  DEBUG:modbus-proxy.ModBus(192.168.1.8:1502):received b'\x01\xd9\x00\x00\x00k~\x03h\x00g\x002\x00\x00\x00\x00\x00\x00\x00\x00\xff\xfe\x0fF\x0f^\x0fZ\x08\xd1\x08\xd3\x08\xe9\xff\xff\x00\x00\x00\x00\x13\x85\xff\xfe\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x96\x81\xbe\x00\x00\xff\xff\x80\x00\x00\x00\xff\xff\x00\x00\x00\x00\x80\x00\x00\x00\x80\x00\x80\x00\xff\xfe\x00\x02\x00\x00\xff\xff\xff\xff\xff\xff\xff\xff\x00\x00\x00\x00\xff\xff\xff\xff\xff\xff\xff\xff\x00\x00\x00\x00',
2021-01-25T22:54:22.765017000Z 2021-01-25 22:54:22,764:  DEBUG:modbus-proxy.Client(192.168.1.87:46806):sending b'\x01\xd9\x00\x00\x00k~\x03h\x00g\x002\x00\x00\x00\x00\x00\x00\x00\x00\xff\xfe\x0fF\x0f^\x0fZ\x08\xd1\x08\xd3\x08\xe9\xff\xff\x00\x00\x00\x00\x13\x85\xff\xfe\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x96\x81\xbe\x00\x00\xff\xff\x80\x00\x00\x00\xff\xff\x00\x00\x00\x00\x80\x00\x00\x00\x80\x00\x80\x00\xff\xfe\x00\x02\x00\x00\xff\xff\xff\xff\xff\xff\xff\xff\x00\x00\x00\x00\xff\xff\xff\xff\xff\xff\xff\xff\x00\x00\x00\x00',
2021-01-25T22:54:23.599594000Z 2021-01-25 22:54:23,599:  DEBUG:modbus-proxy.Client(172.19.0.1:46402):received b'\xb78\x00\x00\x00\x06~\x03\x9c\xfc\x00k',
2021-01-25T22:54:23.600224000Z 2021-01-25 22:54:23,599:  DEBUG:modbus-proxy.ModBus(192.168.1.8:1502):sending b'\xb78\x00\x00\x00\x06~\x03\x9c\xfc\x00k',
2021-01-25T22:54:23.618326000Z 2021-01-25 22:54:23,617:  DEBUG:modbus-proxy.ModBus(192.168.1.8:1502):received b'\xb78\x00\x00\x00\xd9~\x03\xd6\x00\xcb\x00i\x00<\x00\x10\x00\x1a\x00\x11\xff\xffX\xd3X\xd3W\x8aX[\x00\x00\x00\x00\x00\x00\x00\x00\xff\xfe\x13\x87\xff\xfe\xfc#\xfe\xc4\xfe\x0c\xffS\x00\x00\x04\xeb\x01y\x02>\x01Q\x00\x00\xfc\xf4\xff1\xfe\xe5\xfe\xde\x00\x00\x1c\xe8 \xcf!\xfc\x13\xeb\xff\xfe\x001e\xc8\x00\x0e3\x96\x00\x11\xcf\xec\x00\x13\xf4*\x00k-\x07\x006\xa8X\x00$\x11\xd4\x00\x13\x04f\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x80\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x80\x00\x00\x00\x00\x00',
2021-01-25T22:54:23.619193000Z 2021-01-25 22:54:23,618:  DEBUG:modbus-proxy.Client(172.19.0.1:46402):sending b'\xb78\x00\x00\x00\xd9~\x03\xd6\x00\xcb\x00i\x00<\x00\x10\x00\x1a\x00\x11\xff\xffX\xd3X\xd3W\x8aX[\x00\x00\x00\x00\x00\x00\x00\x00\xff\xfe\x13\x87\xff\xfe\xfc#\xfe\xc4\xfe\x0c\xffS\x00\x00\x04\xeb\x01y\x02>\x01Q\x00\x00\xfc\xf4\xff1\xfe\xe5\xfe\xde\x00\x00\x1c\xe8 \xcf!\xfc\x13\xeb\xff\xfe\x001e\xc8\x00\x0e3\x96\x00\x11\xcf\xec\x00\x13\xf4*\x00k-\x07\x006\xa8X\x00$\x11\xd4\x00\x13\x04f\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x80\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x80\x00\x00\x00\x00\x00',
2021-01-25T22:54:23.644884000Z 2021-01-25 22:54:23,644:  DEBUG:modbus-proxy.Client(172.19.0.1:46402):received b"\xb2\xb2\x00\x00\x00\x06~\x03\x9c\x85\x00'",
2021-01-25T22:54:23.645250000Z 2021-01-25 22:54:23,644:  DEBUG:modbus-proxy.ModBus(192.168.1.8:1502):sending b"\xb2\xb2\x00\x00\x00\x06~\x03\x9c\x85\x00'",
2021-01-25T22:54:23.675535000Z 2021-01-25 22:54:23,675:  DEBUG:modbus-proxy.ModBus(192.168.1.8:1502):received b'\xb2\xb2\x00\x00\x00Q~\x03N\x00g\x002\x00\x00\x00\x00\x00\x00\x00\x00\xff\xfe\x0fF\x0f\\\x0f`\x08\xd1\x08\xd3\x08\xee\xff\xff\x00\x00\x00\x00\x13\x85\xff\xfe\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x96\x81\xbe\x00\x00\xff\xff\x80\x00\x00\x00\xff\xf [... cut ...]
2021-01-25T22:54:24.835543000Z 2021-01-25 22:54:24,834:  DEBUG:modbus-proxy.Client(192.168.1.87:46806):sending b'\x01\xdb\x00\x00\x00k~\x03h\x00g\x002\x00\x00\x00\x00\x00\x00\x00\x00\xff\xfe\x0fB\x0fY\x0fZ\x08\xd0\x08\xd0\x08\xe9\xff\xff\x00\x00\x00\x00\x13\x85\xff\xfe\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x96\x81\xbe\x00\x00\xff\xff\x80\x00\x00\x00\xff\xff\x00\x00\x00\x00\x80\x00\x00\x00\x80\x00\x80\x00\xff\xfe\x00\x02\x00\x00\xff\xff\xff\xff\xff\xff\xff\xff\x00\x00\x00\x00\xff\xff\xff\xff\xff\xff\xff\xff\x00\x00\x00\x00',

magpern, 2021-01-26:
Update. It has been running for 10 hours now and is still kicking!

tiagocoutinho, 2021-01-26:
Great! That's excellent news.
I suspect the most important fix was the one that makes sure that the full modbus message is received before sending it.

Concerning logging to a file, IMHO, the most flexible way would be to support a command line argument that is a python logging configuration file (as in [logging.config](https://docs.python.org/3/library/logging.config.html)). 
I guess that depending on the contents of the logging configuration, a matching docker volume would have to be mounted.
I'll have a look at it.

magpern, 2021-01-26:
It's been running for 24 hours now, and till running. This has never been done before.
I have seen some errors in the trace file, but it recuperates from the errors and continues.
I, personally, consider this bug fixed. Gold star to you for your work and extremely fast action. modbis-proxy will be a key component in my production environment for my battery for my house.
I see you have implemented logging to file, I will test that too.
But as far as this issue, I will close it!

2021-01-26 21:30:51,840:  DEBUG:modbus-proxy.ModBus(192.168.1.8:1502):received b'ML\x00\x00\x00k~\x03h\x00g\x002\x00\x00\x00\x00\x00\x00\x00\x00\xff\xfe\x0fn\x0f\x88\x0f\x87\x08\xe6\x08\xe8\t\x07\xff\xff\x00\x00\x00\x00\x13\x84\xff\xfe\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x96\x81\xbe\x00\x00\xff\xff\x80\x00\x00\x00\xff\xff\x00\x00\x00\x00\x80\x00\x00\x00\x80\x00\x80\x00\xff\xfe\x00\x02\x00\x00\xff\xff\xff\xff\xff\xff\xff\xff\x00\x00\x00\x00\xff\xff\xff\xff\xff\xff\xff\xff\x00\x00\x00\x00',
2021-01-26 21:30:51,841:  DEBUG:modbus-proxy.Client(192.168.1.87:54480):sending b'ML\x00\x00\x00k~\x03h\x00g\x002\x00\x00\x00\x00\x00\x00\x00\x00\xff\xfe\x0fn\x0f\x88\x0f\x87\x08\xe6\x08\xe8\t\x07\xff\xff\x00\x00\x00\x00\x13\x84\xff\xfe\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x96\x81\xbe\x00\x00\xff\xff\x80\x00\x00\x00\xff\xff\x00\x00\x00\x00\x80\x00\x00\x00\x80\x00\x80\x00\xff\xfe\x00\x02\x00\x00\xff\xff\xff\xff\xff\xff\xff\xff\x00\x00\x00\x00\xff\xff\xff\xff\xff\xff\xff\xff\x00\x00\x00\x00',
2021-01-26 21:30:52,380:   INFO:modbus-proxy.Client(172.19.0.1:53630):new client connection,
2021-01-26 21:30:52,381:  DEBUG:modbus-proxy.Client(172.19.0.1:53630):received b'\x00\x7f\x00\x00\x00\x06+\x03\x9c@\x00\x02',
2021-01-26 21:30:52,381:  DEBUG:modbus-proxy.ModBus(192.168.1.8:1502):sending b'\x00\x7f\x00\x00\x00\x06+\x03\x9c@\x00\x02',
2021-01-26 21:30:52,561:  DEBUG:modbus-proxy.Client(172.19.0.1:53614):received b'\x85\x1c\x00\x00\x00\x06~\x03\x9c\xfc\x00k',
2021-01-26 21:30:52,844:  DEBUG:modbus-proxy.Client(192.168.1.87:54480):received b'MM\x00\x00\x00\x06~\x03\x9c\x85\x004',
2021-01-26 21:31:02,383:  ERROR:modbus-proxy.ModBus(192.168.1.8:1502):write_read error: TimeoutError(),
2021-01-26 21:31:02,383:   INFO:modbus-proxy.ModBus(192.168.1.8:1502):closing connection...,
2021-01-26 21:31:02,384:   INFO:modbus-proxy.ModBus(192.168.1.8:1502):connection closed,
2021-01-26 21:31:02,384:   INFO:modbus-proxy.ModBus(192.168.1.8:1502):connecting to modbus...,
2021-01-26 21:31:02,389:  ERROR:modbus-proxy.ModBus(192.168.1.8:1502):write_read error: ConnectionRefusedError(111, "Connect call failed ('192.168.1.8', 1502)"),
2021-01-26 21:31:02,389:   INFO:modbus-proxy.Client(172.19.0.1:53630):closing connection...,
2021-01-26 21:31:02,390:   INFO:modbus-proxy.ModBus(192.168.1.8:1502):connecting to modbus...,
2021-01-26 21:31:02,391:   INFO:modbus-proxy.Client(172.19.0.1:53630):connection closed,
2021-01-26 21:31:02,391:  ERROR:modbus-proxy.ModBus(192.168.1.8:1502):write_read error: ConnectionRefusedError(111, "Connect call failed ('192.168.1.8', 1502)"),
2021-01-26 21:31:02,392:   INFO:modbus-proxy.ModBus(192.168.1.8:1502):connecting to modbus...,
2021-01-26 21:31:02,393:  ERROR:modbus-proxy.ModBus(192.168.1.8:1502):write_read error: ConnectionRefusedError(111, "Connect call failed ('192.168.1.8', 1502)"),
2021-01-26 21:31:02,393:   INFO:modbus-proxy.Client(172.19.0.1:53614):closing connection...,
2021-01-26 21:31:02,394:   INFO:modbus-proxy.ModBus(192.168.1.8:1502):connecting to modbus...,
2021-01-26 21:31:02,395:   INFO:modbus-proxy.Client(172.19.0.1:53614):connection closed,
2021-01-26 21:31:02,395:  ERROR:modbus-proxy.ModBus(192.168.1.8:1502):write_read error: ConnectionRefusedError(111, "Connect call failed ('192.168.1.8', 1502)"),
2021-01-26 21:31:02,395:   INFO:modbus-proxy.ModBus(192.168.1.8:1502):connecting to modbus...,
2021-01-26 21:31:02,397:  ERROR:modbus-proxy.ModBus(192.168.1.8:1502):write_read error: ConnectionRefusedError(111, "Connect call failed ('192.168.1.8', 1502)"),
2021-01-26 21:31:02,397:   INFO:modbus-proxy.Client(192.168.1.87:54480):closing connection...,
2021-01-26 21:31:02,397:   INFO:modbus-proxy.Client(192.168.1.87:54480):connection closed,
2021-01-26 21:31:02,399:   INFO:modbus-proxy.Client(172.19.0.1:53650):new client connection,
2021-01-26 21:31:02,399:  DEBUG:modbus-proxy.Client(172.19.0.1:53650):received b"\xc7\xf3\x00\x00\x00\x06~\x03\x9c\x85\x00'",
2021-01-26 21:31:02,400:   INFO:modbus-proxy.ModBus(192.168.1.8:1502):connecting to modbus...,
2021-01-26 21:31:02,400:  ERROR:modbus-proxy.ModBus(192.168.1.8:1502):write_read error: ConnectionRefusedError(111, "Connect call failed ('192.168.1.8', 1502)"),
2021-01-26 21:31:02,400:   INFO:modbus-proxy.ModBus(192.168.1.8:1502):connecting to modbus...,
2021-01-26 21:31:02,401:  ERROR:modbus-proxy.ModBus(192.168.1.8:1502):write_read error: ConnectionRefusedError(111, "Connect call failed ('192.168.1.8', 1502)"),
2021-01-26 21:31:02,401:   INFO:modbus-proxy.Client(172.19.0.1:53650):closing connection...,
2021-01-26 21:31:02,402:   INFO:modbus-proxy.Client(172.19.0.1:53650):connection closed,
2021-01-26 21:31:05,403:   INFO:modbus-proxy.Client(172.19.0.1:53660):new client connection,
2021-01-26 21:31:05,403:  DEBUG:modbus-proxy.Client(172.19.0.1:53660):received b't\x9c\x00\x00\x00\x06~\x03\x9c\xfc\x00k',
2021-01-26 21:31:05,404:   INFO:modbus-proxy.ModBus(192.168.1.8:1502):connecting to modbus...,
[... cut ...] 00\x11\xcf\xec\x00\x13\xf44\x00k\xdc\xbb\x006\xf0\x1a\x00$W \x00\x13\'\x0c\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x80\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x80\x00\x00\x00\x00\x00',

closed 2021-01-26 by magpern

tiagocoutinho, 2021-01-27:
great!
thanks for your kind words.
they help keeping the motivation
i will have a quick look later on at the latest log you sent just to check if the ConnectionRefusedError is due to an error in the proxy.

best of luck with your interesting project

magpern, 2021-09-01:
Here's just a follow up! I've been using this since january 2021. It has not failed one single time, and it is running 24/7.
Super stable!

tiagocoutinho, 2021-09-07:
I'm very glad to hear that it working well.
Thanks a lot for your feedback @magpern!