tiagocoutinho/modbus-proxy #16: Crashing very often

pczolee, 2021-12-10:
* ModBus TCP proxy version: latest
* Python version: 3.5.3
* Operating System: Debian 9.13

### Description

It crashing very often. Now I made a script to start it again, so you can see how often starting it again because of crash:
2021-12-08 20:21:16,260     INFO modbus-proxy: Starting...
2021-12-09 04:02:58,291     INFO modbus-proxy: Starting...
2021-12-09 11:22:36,333     INFO modbus-proxy: Starting...
2021-12-09 11:23:09,014     INFO modbus-proxy: Starting...
2021-12-09 11:23:43,948     INFO modbus-proxy: Starting...
2021-12-09 11:23:53,006     INFO modbus-proxy: Starting...
2021-12-09 11:24:32,073     INFO modbus-proxy: Starting...
2021-12-09 11:25:01,598     INFO modbus-proxy: Starting...
2021-12-09 11:26:02,074     INFO modbus-proxy: Starting...
2021-12-09 11:26:10,315     INFO modbus-proxy: Starting...
2021-12-09 11:26:19,340     INFO modbus-proxy: Starting...
2021-12-09 11:27:28,007     INFO modbus-proxy: Starting...
2021-12-09 11:28:02,084     INFO modbus-proxy: Starting...
2021-12-09 11:28:36,598     INFO modbus-proxy: Starting...
2021-12-09 11:28:45,659     INFO modbus-proxy: Starting...
2021-12-09 11:29:32,091     INFO modbus-proxy: Starting...
2021-12-09 11:29:54,347     INFO modbus-proxy: Starting...
2021-12-09 11:31:02,118     INFO modbus-proxy: Starting...
2021-12-09 11:31:08,986     INFO modbus-proxy: Starting...
2021-12-09 11:32:17,690     INFO modbus-proxy: Starting...
2021-12-09 11:32:26,692     INFO modbus-proxy: Starting...
2021-12-09 11:32:29,820     INFO modbus-proxy: Starting...
2021-12-09 11:33:32,377     INFO modbus-proxy: Starting...
2021-12-09 11:34:02,104     INFO modbus-proxy: Starting...
2021-12-09 11:34:40,983     INFO modbus-proxy: Starting...
2021-12-09 11:34:50,063     INFO modbus-proxy: Starting...
2021-12-09 11:35:32,090     INFO modbus-proxy: Starting...
2021-12-09 11:35:58,748     INFO modbus-proxy: Starting...
2021-12-09 11:37:02,109     INFO modbus-proxy: Starting...
2021-12-09 11:37:05,172     INFO modbus-proxy: Starting...
2021-12-09 11:37:13,413     INFO modbus-proxy: Starting...
2021-12-09 11:38:22,082     INFO modbus-proxy: Starting...
2021-12-09 11:38:31,046     INFO modbus-proxy: Starting...
2021-12-09 11:39:36,747     INFO modbus-proxy: Starting...
2021-12-09 11:40:02,162     INFO modbus-proxy: Starting...
2021-12-09 11:40:45,386     INFO modbus-proxy: Starting...
2021-12-09 11:40:54,435     INFO modbus-proxy: Starting...
2021-12-09 11:41:32,117     INFO modbus-proxy: Starting...
2021-12-09 11:42:03,054     INFO modbus-proxy: Starting...
2021-12-09 11:43:02,114     INFO modbus-proxy: Starting...
2021-12-09 11:43:11,722     INFO modbus-proxy: Starting...
2021-12-09 11:43:20,764     INFO modbus-proxy: Starting...
2021-12-09 11:44:29,422     INFO modbus-proxy: Starting...
2021-12-09 11:45:38,015     INFO modbus-proxy: Starting...
2021-12-09 11:45:47,085     INFO modbus-proxy: Starting...
2021-12-09 11:46:02,148     INFO modbus-proxy: Starting...
2021-12-09 11:46:55,756     INFO modbus-proxy: Starting...
2021-12-09 11:47:32,133     INFO modbus-proxy: Starting...
2021-12-09 11:48:04,427     INFO modbus-proxy: Starting...
2021-12-09 11:48:13,454     INFO modbus-proxy: Starting...
2021-12-09 11:49:02,196     INFO modbus-proxy: Starting...
2021-12-09 11:49:22,140     INFO modbus-proxy: Starting...
2021-12-09 11:50:30,799     INFO modbus-proxy: Starting...
2021-12-09 11:50:39,766     INFO modbus-proxy: Starting...
2021-12-09 11:51:48,471     INFO modbus-proxy: Starting...
2021-12-09 11:52:02,148     INFO modbus-proxy: Starting...
2021-12-09 11:52:57,093     INFO modbus-proxy: Starting...
2021-12-09 11:53:06,117     INFO modbus-proxy: Starting...
2021-12-09 11:53:32,206     INFO modbus-proxy: Starting...
2021-12-09 11:54:14,733     INFO modbus-proxy: Starting...
2021-12-09 11:55:02,164     INFO modbus-proxy: Starting...
2021-12-09 11:55:23,488     INFO modbus-proxy: Starting...
2021-12-09 11:55:32,483     INFO modbus-proxy: Starting...
[... 146 lines cut ...]
2021-12-09 04:02:27,129    DEBUG modbus-proxy.Client(127.0.0.1:48351): sending b"\x02\xbd\x00\x00\x00O\x01\x03L\x00\x00\x00\x00\x00\x00\x00\x00\xff\xfe\x0f\xd9\x0f\xd1\x0f\xd9\t5\t\x12\t'\xff\xff\x00\x00\x00\x00\x13\x89\xff\xfe\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02&\xa1\xe4\x00\x00\xff\xff\x80\x00\x00\x00\xff\xff\x00\x00\x00\x00\x80\x00\x00\x00\x80\x00\x80\x00\xff\xfe\x00\x02\x00\x00"
2021-12-09 04:02:57,103    DEBUG modbus-proxy.Client(127.0.0.1:48351): received b'\x02\xbe\x00\x00\x00\x06\x01\x03\x9c\x87\x00&'
2021-12-09 04:02:57,103    DEBUG modbus-proxy.ModBus(192.168.1.49:1502): sending b'\x02\xbe\x00\x00\x00\x06\x01\x03\x9c\x87\x00&'
2021-12-09 04:02:57,116    ERROR modbus-proxy.ModBus(192.168.1.49:1502): write_read error: ConnectionResetError(104, 'Connection reset by peer')
2021-12-09 04:02:57,116     INFO modbus-proxy.ModBus(192.168.1.49:1502): closing connection...
2021-12-09 04:02:57,116     INFO modbus-proxy.ModBus(192.168.1.49:1502): failed to close: ConnectionResetError(104, 'Connection reset by peer')
2021-12-09 04:02:57,117     INFO modbus-proxy.ModBus(192.168.1.49:1502): connecting to modbus...
2021-12-09 04:02:57,117    ERROR modbus-proxy.ModBus(192.168.1.49:1502): write_read error: CancelledError()
2021-12-09 04:02:57,118     INFO modbus-proxy.Client(127.0.0.1:48351): closing connection...
2021-12-09 04:02:57,118    ERROR modbus-proxy.Client(127.0.0.1:39943): reading error: CancelledError()
2021-12-09 04:02:57,118     INFO modbus-proxy.Client(127.0.0.1:39943): closing connection...
2021-12-09 04:02:57,118     INFO modbus-proxy.Client(127.0.0.1:48351): connection closed
2021-12-09 04:02:57,118     INFO modbus-proxy.Client(127.0.0.1:39943): connection closed
Traceback (most recent call last):
  File "/usr/local/bin/modbus-proxy", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.7/site-packages/modbus_proxy.py", line 343, in main
    asyncio.run(run())
  File "/usr/local/lib/python3.7/asyncio/runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "/usr/local/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
    return future.result()
concurrent.futures._base.CancelledError

Thanks

tiagocoutinho, 2021-12-17:
Thanks for the detailed report.

Could this be related with #4?

Try launching with a delay between connection and first request. Example:

```bash
$ modbus-proxy -b tcp://0:5020 --modbus tcp://192.168.1.49:1502 --modbus-connection-time=2
```

pczolee, 2021-12-17:
Thanks for the answer. My config is now:

> devices:
- modbus:
    url: 192.168.1.49:1502     # device url (mandatory)
    timeout: 10                # communication timeout (s) (optional, default: 10)
    connection_time: 0.1       # delay after connection (s) (optional, default: 0)
  listen:
    bind: 0:502               # listening address (mandatory)
logging:
  version: 1
  formatters:
    standard:
      format: "%(asctime)s %(levelname)8s %(name)s: %(message)s"
  handlers:
    console:
      class: logging.StreamHandler
      formatter: standard
  root:
    handlers: ['console']
    level: DEBUG

So I changed connection_time: 0.1 to connection_time: 2 . Let's ses if it is helping ;) .

pczolee, 2021-12-17:
Unfortunately I still have the problem :(

tiagocoutinho, 2021-12-18:
I think I have a fix for your issue.
I just made a new release 0.6.8. Could you test with it (or if you prefer the latest master from git)?
Thanks in advance

pczolee, 2021-12-18:
Ohh, sounds good! Thanks.
Well, I would like to try your new version, but unfortunately the pip still have just 0.6.7 . Can you tell me how can I use the master from it?

pczolee, 2021-12-19:
The update went well today, so now the 0.6.8 is running. I'll let you know after a day of test.
Thanks

pczolee, 2021-12-20:
I think you found the problem, because there was no more crash since the program was last launched (~21 hours ago).
Thanks for your help, and for this great program!

tiagocoutinho, 2021-12-20:
Great! glad to hear that it works.
Thanks for the feedback and the detailed report. It helps to improve the project.

commit b0cfbf15 (2021-12-18): Fix #16: Crashing very often (#17)

* Better logging attempts when communicating with modbus

* Fix #16: Crashing very often

When an error occurs with modbus don't stop the server.
Instead, simply close the connection to modbus

* Remove temporarely pypy github tests
closed 2021-12-20 by tiagocoutinho