tiagocoutinho/modbus-proxy #18: Thank you! -  and HomeAssistant Addon

Akulatraxas, 2022-04-17:
Just wanted to leave a thank you here for creating this. It was exactly what i needed. I am running homeassistant and a solaredge inverter. The Wallbox and the home assistant both want to talk to the inverter and this was impossible.
I created a homeassistant addon (https://www.home-assistant.io/addons/) for this and was asked to make it public so others can use it as well . 

Its the first time for me doing something like this and i am unsure how to attribute your work. I added a Mention and the Link back to this repository - is there a more appropriate way to do it?

https://github.com/Akulatraxas/ha-modbusproxy

ChristophCaina, 2022-05-25:
thank you, @Akulatraxas 
Until last week, I had my SolarEdge Inverter integrated into HomeAssistant by using the SolarEdge Modbus Integration.
Last week, I got a wallbox installed - so I had the exact same issue as you mentioned above.

Now, I found this modbus-proxy, which could solve the connection issues - but wasn't sure, if I can run it within the Addon "pyscript" for HomeAssistant - or if I should create a dedicated machine in my VM environment for this purpose.

Now, after seeing your comment - I can throw away all my concerns and could run your ha-modbusproxy instead :)

tiagocoutinho, 2022-05-26:
Thank you for your @Akulatraxas.

Thank you also for your concern in making sure proper credit is given to this project.
Since I don't know home assistant and its addons/plugins system, I cannot comment on the details of your project.
As far as I'm concerned, what you did concerning credit to my work is appropriate.

closed 2022-05-26 by tiagocoutinho