tiagocoutinho/modbus-proxy #19: Systemd service

ChristophCaina, 2022-10-15:
It would make sense to give a short description on how to setup the modbus proxy as a service (systemd)...
I don't know, if this should go to the "readme" directly, since it might be different depending on the host system... therefore, it might be something for a wiki instead?
Therefore I haven't added this to the readme and made a pr... if you want, I can add it :)

1. I have moved the config file to `/usr/lib/mproxy-conf.yaml`
2. go to `/etc/systemd/system/`
3. use nano to create a service file `nano mproxy.service`
4. the file should contain the following information:
```
[Unit]
Description=Modbus-Proxy
After=network.target

[Service]
Type=simple
Restart=always
ExecStart = modbus-proxy -c ./usr/lib/mproxy-conf.yaml

[Install]
WantedBy=multi-user.target
```
5. `run systemctl daemon-reload`
6. `systemctl enable mproxy.service`
7. `systemctl start mproxy.service`

With the command systemctl status mproxy.service can you check, if the software / service was started successfully and if it is running.

tiagocoutinho, 2022-11-25:
Thank you for the proposal.
I think this could be a good chapter for the readme.
If you could make a PR it would be awesome

ChristophCaina, 2022-11-25:
done #21

closed 2022-11-27 by tiagocoutinho