tiagocoutinho/modbus-proxy #22: Systemd service does not work as described

winfriedp, 2022-12-15:
* ModBus TCP proxy version: 0.6.8
* Python version: 3.9
* Operating System: Raspberry Pi OS Bullseye v11.5

### Description

The service configuration does not work when installing modbus-proxy according to the instructions. I get an ModuleNotFoundError: No module named 'modbus_proxy'. 
The reason is that modbus-proxy is installed in a local user directory, but the service is run by the user root and root does not know the local directory. I am not a python expert and I have not found a solution to adjust the environment variables so that root can handle them.

### What I Did

I solved the problem by installing modbus-proxy as user root as follows:
  sudo su
  pip3 install modbus-proxy

Then the package will be installed in /usr/local/bin/ and /usr/local/lib/python3.9/dist-packages
Now the service can be started without error. This workaround is OK for me, but maybe there are other suggestions from python experts to make this service work.

requiemmg, 2023-01-14:
I cannot start the service. First I did as stated in the documentation, status failed.
Then I saw your report and uninstalled modbus-proxy as a user and reinstalled as root.
I also tried to use absolute paths, but no success.
When I run from bash, everything is fine.

`● mproxy.service                                              loaded failed     failed        Modbus-Proxy`

My mproxy.service (in etc/system/systemd) looks like this:

`[Unit]
Description=Modbus-Proxy
After=network.target

[Service]
Type=simple
Restart=always
ExecStart = /usr/local/bin/modbus-proxy -c /usr/lib/mproxy-conf.yaml

[Install]
WantedBy=multi-user.target`

Any ideas?

winfriedp, 2023-01-14:
The only difference to my service definition is the ExecStart command.
Try
`ExecStart = modbus-proxy -c ./usr/lib/mproxy-conf.yaml`
I guess it's only the missing dot before the slash.

requiemmg, 2023-01-15:
Did so, still failing and I dont know why as it runs smoothly from the command line.

tiagocoutinho, 2023-01-15:
Could you send the systemctl log:

```
journalctl -u mproxy > /tmp/mproxy-log.txt
```

requiemmg, 2023-01-15:
Sure, no problem. I have it working now with a modbus-proxy fork for Home Assistant.

[mproxy.log.txt](https://github.com/tiagocoutinho/modbus-proxy/files/10421149/mproxy.log.txt)

tiagocoutinho, 2023-01-15:
looks like you installed modbus-proxy on root without yaml support. Maybe you did something like:
```sudo pip install modbus-proxy```

You need to do instead:
```sudo pip install modbus-proxy[yaml]```

requiemmg, 2023-01-15:
Thanks for the clarification!

thaichho, 2024-01-16:
I had also issues but with the help of this thread I was able to solve them.
Anyway it might be better do enhance the README File

Richigeht, 2024-02-12:
@tiagocoutinho why don't you do that with a PR on this repo and more information? :-)

tiagocoutinho, 2025-06-08:
@AndrCarl added systemd example to the Readme.
Closing

closed 2025-06-08 by tiagocoutinho