tiagocoutinho/modbus-proxy #24: Not possible to connect more than one client

winfriedp, 2022-12-27:
* ModBus TCP proxy version: 0.6.8
* Python version: 3.9
* Operating System: Raspberry Pi OS Bullseye v11.5

### Description
Two master clients (FHEM and CCGX) are to connect with the Solaredge inverter over modbus-proxy, but only one does.
Either one or the other, but not both. One client connection results always in a ConnectionRefusedError.
I have tried different combinations of connection_time and timeout values without success.

### What I Did
This is my mproxy-conf.yaml:
`devices:
- modbus:
    url: 192.168.66.57:502     # solaredge inverter url (mandatory)
    timeout: 10                # communication timeout (s) (optional, default: 10)
    connection_time: 0.1       # delay after connection (s) (optional, default: 0)
  listen:
    bind: 0:9000               # listening address (for FHEM)
- modbus:
    url: 192.168.66.57:502     # solaredge inverter url
    timeout: 10
    connection_time: 0.1
  listen:
    bind: 0:502                # listening address (for CCGX, requires port 502)
logging:
  version: 1
  formatters:
    standard:
      format: "%(asctime)s %(levelname)8s %(name)s: %(message)s"
  handlers:
    console:
      class: logging.StreamHandler
      formatter: standard
  root:
    handlers: ['console']
    level: DEBUG`

Here the syslog output:
`Dec 27 21:13:03 rsp2 systemd[1]: Started Modbus-Proxy.
Dec 27 21:13:04 rsp2 modbus-proxy[13616]: 2022-12-27 21:13:04,113     INFO modbus-proxy: Starting...
Dec 27 21:13:04 rsp2 modbus-proxy[13616]: 2022-12-27 21:13:04,121     INFO modbus-proxy.ModBus(192.168.66.57:502): Ready to accept requests on 0:9000
Dec 27 21:13:04 rsp2 modbus-proxy[13616]: 2022-12-27 21:13:04,122     INFO modbus-proxy.ModBus(192.168.66.57:502): Ready to accept requests on 0:502
Dec 27 21:13:16 rsp2 modbus-proxy[13616]: 2022-12-27 21:13:16,750     INFO modbus-proxy.Client(192.168.66.65:39376): new client connection
Dec 27 21:13:16 rsp2 modbus-proxy[13616]: 2022-12-27 21:13:16,876    DEBUG modbus-proxy.Client(192.168.66.65:39376): received b'\x00^\x00\x00\x00\x06~\x03\x9cl\x00\x08'
Dec 27 21:13:16 rsp2 modbus-proxy[13616]: 2022-12-27 21:13:16,877     INFO modbus-proxy.ModBus(192.168.66.57:502): connecting to modbus...
Dec 27 21:13:16 rsp2 modbus-proxy[13616]: 2022-12-27 21:13:16,885     INFO modbus-proxy.ModBus(192.168.66.57:502): connected!
Dec 27 21:13:16 rsp2 modbus-proxy[13616]: 2022-12-27 21:13:16,886     INFO modbus-proxy.ModBus(192.168.66.57:502): delay after connect: 0.1
Dec 27 21:13:16 rsp2 modbus-proxy[13616]: 2022-12-27 21:13:16,987    DEBUG modbus-proxy.ModBus(192.168.66.57:502): sending b'\x00^\x00\x00\x00\x06~\x03\x9cl\x00\x08'
Dec 27 21:13:17 rsp2 modbus-proxy[13616]: 2022-12-27 21:13:17,001    DEBUG modbus-proxy.ModBus(192.168.66.57:502): received b'\x00^\x00\x00\x00\x13~\x03\x100004.0016.0025\x00\x00'
Dec 27 21:13:17 rsp2 modbus-proxy[13616]: 2022-12-27 21:13:17,002    DEBUG modbus-proxy.Client(192.168.66.65:39376): sending b'\x00^\x00\x00\x00\x13~\x03\x100004.0016.0025\x00\x00'
Dec 27 21:13:17 rsp2 modbus-proxy[13616]: 2022-12-27 21:13:17,179    DEBUG modbus-proxy.Client(192.168.66.65:39376): received b'\x00W\x00\x00\x00\x06~\x03\x9c\x85\x00\x01'
Dec 27 21:13:17 rsp2 modbus-proxy[13616]: 2022-12-27 21:13:17,180    DEBUG modbus-proxy.ModBus(192.168.66.57:502): sending b'\x00W\x00\x00\x00\x06~\x03\x9c\x85\x00\x01'
Dec 27 21:13:17 rsp2 modbus-proxy[13616]: 2022-12-27 21:13:17,203    DEBUG modbus-proxy.ModBus(192.168.66.57:502): received b'\x00W\x00\x00\x00\x05~\x03\x02\x00g'
Dec 27 21:13:17 rsp2 modbus-proxy[13616]: 2022-12-27 21:13:17,203    DEBUG modbus-proxy.Client(192.168.66.65:39376): sending b'\x00W\x00\x00\x00\x05~\x03\x02\x00g'
Dec 27 21:13:17 rsp2 modbus-proxy[13616]: 2022-12-27 21:13:17,308    DEBUG modbus-proxy.Client(192.168.66.65:39376): received b'\x00?\x00\x00\x00\x06~\x03\x9c\x87\x00\x01'
Dec 27 21:13:17 rsp2 modbus-proxy[13616]: 2022-12-27 21:13:17,309    DEBUG modbus-proxy.ModBus(192.168.66.57:502): sending b'\x00?\x00\x00\x00\x06~\x03\x9c\x87\x00\x01'
Dec 27 21:13:17 rsp2 modbus-proxy[13616]: 2022-12-27 21:13:17,333    DEBUG modbus-proxy.ModBus(192.168.66.57:502): received b'\x00?\x00\x00\x00\x05~\x03\x02\x00\x00'
Dec 27 21:13:17 rsp2 modbus-proxy[13616]: 2022-12-27 21:13:17,334    DEBUG modbus-proxy.Client(192.168.66.65:39376): sending b'\x00?\x00\x00\x00\x05~\x03\x02\x00\x00'
Dec 27 21:13:17 rsp2 modbus-proxy[13616]: 2022-12-27 21:13:17,438    DEBUG modbus-proxy.Client(192.168.66.65:39376): received b'\x00\xa7\x00\x00\x00\x06~\x03\x9c\x88\x00\x01'
Dec 27 21:13:17 rsp2 modbus-proxy[13616]: 2022-12-27 21:13:17,439    DEBUG modbus-proxy.ModBus(192.168.66.57:502): sending b'\x00\xa7\x00\x00\x00\x06~\x03\x9c\x88\x00\x01'
Dec 27 21:13:17 rsp2 modbus-proxy[13616]: 2022-12-27 21:13:17,473    DEBUG modbus-proxy.ModBus(192.168.66.57:502): received b'\x00\xa7\x00\x00\x00\x05~\x03\x02\x00\x00'
Dec 27 21:13:17 rsp2 modbus-proxy[13616]: 2022-12-27 21:13:17,473    DEBUG modbus-proxy.Client(192.168.66.65:39376): sending b'\x00\xa7\x00\x00\x00\x05~\x03\x02\x00\x00'
Dec 27 21:13:17 rsp2 modbus-proxy[13616]: 2022-12-27 21:13:17,579    DEBUG modbus-proxy.Client(192.168.66.65:39376): received b'\x00m\x00\x00\x00\x06~\x03\x9c\x89\x00\x01'
Dec 27 21:13:17 rsp2 modbus-proxy[13616]: 2022-12-27 21:13:17,580    DEBUG modbus-proxy.ModBus(192.168.66.57:502): sending b'\x00m\x00\x00\x00\x06~\x03\x9c\x89\x00\x01'
Dec 27 21:13:17 rsp2 modbus-proxy[13616]: 2022-12-27 21:13:17,601    DEBUG modbus-proxy.ModBus(192.168.66.57:502): received b'\x00m\x00\x00\x00\x05~\x03\x02\x00\x00'
Dec 27 21:13:17 rsp2 modbus-proxy[13616]: 2022-12-27 21:13:17,602    DEBUG modbus-proxy.Client(192.168.66.65:39376): sending b'\x00m\x00\x00\x00\x05~\x03\x02\x00\x00'
Dec 27 21:13:17 rsp2 modbus-proxy[13616]: 2022-12-27 21:13:17,716    DEBUG modbus-proxy.Client(192.168.66.65:39376): received b'\x006\x00\x00\x00\x06~\x03\x9c\x8a\x00\x01'
Dec 27 21:13:17 rsp2 modbus-proxy[13616]: 2022-12-27 21:13:17,717    DEBUG modbus-proxy.ModBus(192.168.66.57:502): sending b'\x006\x00\x00\x00\x06~\x03\x9c\x8a\x00\x01'
Dec 27 21:13:17 rsp2 modbus-proxy[13616]: 2022-12-27 21:13:17,733    DEBUG modbus-proxy.ModBus(192.168.66.57:502): received b'\x006\x00\x00\x00\x05~\x03\x02\x00\x00'
Dec 27 21:13:17 rsp2 modbus-proxy[13616]: 2022-12-27 21:13:17,733    DEBUG modbus-proxy.Client(192.168.66.65:39376): sending b'\x006\x00\x00\x00\x05~\x03\x02\x00\x00'
[... 251 lines cut ...]
Dec 27 21:14:06 rsp2 modbus-proxy[13616]: 2022-12-27 21:14:06,131    DEBUG modbus-proxy.Client(192.168.66.65:39376): sending b'\x004\x00\x00\x00\x07~\x03\x04\x00\xa3_\xc9'
Dec 27 21:14:06 rsp2 modbus-proxy[13616]: 2022-12-27 21:14:06,235    DEBUG modbus-proxy.Client(192.168.66.65:39376): received b'\x00`\x00\x00\x00\x06~\x03\x9d2\x00\x01'
Dec 27 21:14:06 rsp2 modbus-proxy[13616]: 2022-12-27 21:14:06,236    DEBUG modbus-proxy.ModBus(192.168.66.57:502): sending b'\x00`\x00\x00\x00\x06~\x03\x9d2\x00\x01'
Dec 27 21:14:06 rsp2 modbus-proxy[13616]: 2022-12-27 21:14:06,270    DEBUG modbus-proxy.ModBus(192.168.66.57:502): received b'\x00`\x00\x00\x00\x05~\x03\x02\x00\x00'
Dec 27 21:14:06 rsp2 modbus-proxy[13616]: 2022-12-27 21:14:06,271    DEBUG modbus-proxy.Client(192.168.66.65:39376): sending b'\x00`\x00\x00\x00\x05~\x03\x02\x00\x00'
Dec 27 21:14:06 rsp2 modbus-proxy[13616]: 2022-12-27 21:14:06,376    DEBUG modbus-proxy.Client(192.168.66.65:39376): received b'\x00r\x00\x00\x00\x06~\x03\x9d3\x00\x02'
Dec 27 21:14:06 rsp2 modbus-proxy[13616]: 2022-12-27 21:14:06,376    DEBUG modbus-proxy.ModBus(192.168.66.57:502): sending b'\x00r\x00\x00\x00\x06~\x03\x9d3\x00\x02'
Dec 27 21:14:06 rsp2 modbus-proxy[13616]: 2022-12-27 21:14:06,390    DEBUG modbus-proxy.ModBus(192.168.66.57:502): received b'\x00r\x00\x00\x00\x07~\x03\x04\x00\x00\x00\x00'
Dec 27 21:14:06 rsp2 modbus-proxy[13616]: 2022-12-27 21:14:06,391    DEBUG modbus-proxy.Client(192.168.66.65:39376): sending b'\x00r\x00\x00\x00\x07~\x03\x04\x00\x00\x00\x00'
Dec 27 21:14:06 rsp2 modbus-proxy[13616]: 2022-12-27 21:14:06,496    DEBUG modbus-proxy.Client(192.168.66.65:39376): received b'\x00\r\x00\x00\x00\x06~\x03\x9dC\x00\x01'
Dec 27 21:14:06 rsp2 modbus-proxy[13616]: 2022-12-27 21:14:06,497    DEBUG modbus-proxy.ModBus(192.168.66.57:502): sending b'\x00\r\x00\x00\x00\x06~\x03\x9dC\x00\x01'
Dec 27 21:14:06 rsp2 modbus-proxy[13616]: 2022-12-27 21:14:06,531    DEBUG modbus-proxy.ModBus(192.168.66.57:502): received b'\x00\r\x00\x00\x00\x05~\x03\x02\x80\x00'
[... cut ...]
Dec 27 21:14:29 rsp2 modbus-proxy[13616]: 2022-12-27 21:14:29,169    ERROR modbus-proxy.ModBus(192.168.66.57:502): write_read error [2/2]: ConnectionRefusedError(111, "Connect call failed ('192.168.66.57', 502)")
Dec 27 21:14:29 rsp2 modbus-proxy[13616]: 2022-12-27 21:14:29,169     INFO modbus-proxy.Client(192.168.66.95:47560): closing connection...
Dec 27 21:14:29 rsp2 modbus-proxy[13616]: 2022-12-27 21:14:29,171     INFO modbus-proxy.Client(192.168.66.95:47560): connection closed
Dec 27 21:14:42 rsp2 systemd[1]: Stopping Modbus-Proxy...
Dec 27 21:14:42 rsp2 systemd[1]: mproxy.service: Succeeded.
Dec 27 21:14:42 rsp2 systemd[1]: Stopped Modbus-Proxy.
`

tiagocoutinho, 2022-12-28:
Declaring 2 modbus devices that connect to the same modbus hardware triggers 2 independent TCP connections. 

A single modbus device supports multiple client connections. Can't the FHEM software connect to the same port as CCGX?

I think the following configuration should be enough:

```yaml
devices:
- modbus:
    url: 192.168.66.57:502     # solaredge inverter url (mandatory)
    timeout: 10                # communication timeout (s) (optional, default: 10)
    connection_time: 0.1       # delay after connection (s) (optional, default: 0)
  listen:
    bind: 0:502               # listening address for both FHEM and CCGX
logging:
  version: 1
  formatters:
    standard:
      format: "%(asctime)s %(levelname)8s %(name)s: %(message)s"
  handlers:
    console:
      class: logging.StreamHandler
      formatter: standard
  root:
    handlers: ['console']
    level: DEBUG`
```

winfriedp, 2022-12-28:
Thanks for your quick response. That was the solution. I misunderstood the configuration. With your config works my setup. I'm happy!

closed 2022-12-28 by winfriedp