tiagocoutinho/modbus-proxy #25: Question: Forwarding slave to another

Gert-dev, 2023-01-27:
Hi

I'm wondering if it is possible to forward one slave index to another, or - in other words - to change its index? For example: ModBus server foo.com:502 has a slave on index 0, and I would like to proxy it, but forward slave **1** to foo.com:502 slave **0** instead of 0 to 0 or 1 to 1.

The reason I'm asking this is because some MODBUS/TCP servers, such as the Huawei SUN-2000L, have their slave on index 0, which - if I understood correctly - is not meant to be used as _actual_ slave index. As a result, some software (e.g. Loxone's MODBUS support) refuses to read from it and requires a minimum slave index of 1. Meanwhile, tools such as qModMaster work fine with slave index 0 to read the necessary data. Having this support would allow me to work around this problem with quirky MODBUS implementations, and link both.

Thanks!

tiagocoutinho, 2023-01-28:
Hi,
So far we only support [modbus TCP protocol](https://en.wikipedia.org/wiki/Modbus#Modbus_TCP_frame_format). So when you say "slave index" are you talking about the "Unit identifier" ?

Assuming you are the answer is no. We do not tampering with any part of the Modbus TCP frame.

I do think it is an interesting feature so if anyone is willing it I would be honored to code review a PR.

Gert-dev, 2023-01-29:
> So far we only support [modbus TCP protocol](https://en.wikipedia.org/wiki/Modbus#Modbus_TCP_frame_format). So when you say "slave index" are you talking about the "Unit identifier" ?

Yes, I think that is indeed it, e.g. what is `253` here:

[image]

(I got confused because there seems to be a bug on qModMaster where if you had it on "TCP" before, on startup it will still say "Slave Address" as is the case for RTU; it fixes itself when you go back and forth.)

> I do think it is an interesting feature so if anyone is willing it I would be honored to code review a PR.

:+1: I can't make any promises, but I might take a whack at this at some point. I'm also investigating alternative solutions simultaneously (i.e. getting Loxone/Huawei to fix their stuff in my case), but I'm pessimistic either will do so.

PR #26 (2024-03-19): Support remapping unit IDs for wonky clients and servers
Some clients require a minimum unit ID of 1 (e.g. Loxone home automation's MODBUS server support), and some servers have a device on unit ID 0 (e.g. Huawei SUN-2000L), making both incompatible. Ideally they should support one another, but when wonky implementations such as these don't, with this PR you can let the proxy remap one unit ID to another to fix this.

Marking this as draft since I've already tested this working with qModMaster, but still need to test on-device to ensure it actually fixes my problem :wink: .

Fixes #25.
closed 2024-03-19 by tiagocoutinho