tiagocoutinho/modbus-proxy #27: Client error: RecvError(()) when connecting to modus-proxy

nicx, 2023-02-05:
I try to run modus-proxy via Docker on my Unraid server. I am using a Huawei SUN2000 10KTL inverter. I want to connect EVCC and Home Assistant simultaneously to modes-proxy. I mapped the container port 502 to the host port 5020.

My config.yaml:
```
devices:
- modbus:
    url: 192.168.1.4:502
    timeout: 10                # communication timeout (s) (optional, default: 10)
    connection_time: 2       # delay after connection (s) (optional, default: 0)
  listen:
    bind: 0:502
```

After starting modus-proxy I tried to connect my clients to it, but with no success. I get a `Client error: RecvError(())` for both clients.

The log output of modus-proxy:
```
[2023-02-05T08:24:13Z INFO  modbus_proxy_rs] Ready to accept requests on 0:502 to 192.168.1.4:502
[2023-02-05T08:25:50Z INFO  modbus_proxy_rs] new client connection (active = 1)
[2023-02-05T08:25:51Z INFO  modbus_proxy_rs] modbus request 192.168.1.4:502: 12 bytes
[2023-02-05T08:25:51Z DEBUG modbus_proxy_rs] modbus request 192.168.1.4:502: [0, 1, 0, 0, 0, 6, 0, 3, 167, 254, 0, 1]
[2023-02-05T08:25:51Z INFO  modbus_proxy_rs] modbus connection to 192.168.1.4:502 sucessfull
[2023-02-05T08:28:51Z INFO  modbus_proxy_rs] new client connection (active = 2)
[2023-02-05T08:28:51Z INFO  modbus_proxy_rs] modbus request 192.168.1.4:502: 12 bytes
[2023-02-05T08:28:51Z DEBUG modbus_proxy_rs] modbus request 192.168.1.4:502: [0, 1, 0, 0, 0, 6, 1, 3, 167, 254, 0, 1]
[2023-02-05T08:28:51Z ERROR modbus_proxy_rs] Client error: RecvError(())
[2023-02-05T08:28:51Z INFO  modbus_proxy_rs] modbus connection to 192.168.1.4:502 sucessfull
[2023-02-05T08:31:52Z INFO  modbus_proxy_rs] new client connection (active = 3)
[2023-02-05T08:31:52Z INFO  modbus_proxy_rs] new client connection (active = 4)
[2023-02-05T08:31:52Z INFO  modbus_proxy_rs] new client connection (active = 5)
[2023-02-05T08:31:52Z INFO  modbus_proxy_rs] modbus request 192.168.1.4:502: 12 bytes
[2023-02-05T08:31:52Z DEBUG modbus_proxy_rs] modbus request 192.168.1.4:502: [0, 1, 0, 0, 0, 6, 1, 3, 125, 64, 0, 2]
[2023-02-05T08:31:52Z ERROR modbus_proxy_rs] Client error: RecvError(())
[2023-02-05T08:31:52Z INFO  modbus_proxy_rs] modbus connection to 192.168.1.4:502 sucessfull
[2023-02-05T08:35:23Z INFO  modbus_proxy_rs] Ready to accept requests on 0:502 to 192.168.1.4:502
[2023-02-05T08:35:24Z INFO  modbus_proxy_rs] new client connection (active = 1)
[2023-02-05T08:35:24Z INFO  modbus_proxy_rs] new client connection (active = 2)
[2023-02-05T08:35:24Z INFO  modbus_proxy_rs] new client connection (active = 3)
[2023-02-05T08:35:24Z INFO  modbus_proxy_rs] new client connection (active = 4)
[2023-02-05T08:35:24Z INFO  modbus_proxy_rs] new client connection (active = 5)
[2023-02-05T08:35:24Z INFO  modbus_proxy_rs] modbus request 192.168.1.4:502: 12 bytes
[2023-02-05T08:35:24Z DEBUG modbus_proxy_rs] modbus request 192.168.1.4:502: [0, 36, 0, 0, 0, 6, 1, 3, 144, 137, 0, 2]
[2023-02-05T08:35:24Z INFO  modbus_proxy_rs] modbus connection to 192.168.1.4:502 sucessfull
[2023-02-05T08:38:25Z INFO  modbus_proxy_rs] new client connection (active = 6)
[2023-02-05T08:38:25Z INFO  modbus_proxy_rs] modbus request 192.168.1.4:502: 12 bytes
[2023-02-05T08:38:25Z DEBUG modbus_proxy_rs] modbus request 192.168.1.4:502: [0, 37, 0, 0, 0, 6, 1, 3, 144, 140, 0, 1]
[2023-02-05T08:38:25Z ERROR modbus_proxy_rs] Client error: RecvError(())
[2023-02-05T08:38:25Z INFO  modbus_proxy_rs] modbus connection to 192.168.1.4:502 sucessfull
[2023-02-05T08:41:26Z INFO  modbus_proxy_rs] new client connection (active = 7)
[2023-02-05T08:41:26Z INFO  modbus_proxy_rs] new client connection (active = 8)
[2023-02-05T08:41:26Z INFO  modbus_proxy_rs] modbus request 192.168.1.4:502: 12 bytes
[2023-02-05T08:41:26Z DEBUG modbus_proxy_rs] modbus request 192.168.1.4:502: [0, 1, 0, 0, 0, 6, 1, 3, 167, 254, 0, 1]
[2023-02-05T08:41:26Z ERROR modbus_proxy_rs] Client error: RecvError(())
[2023-02-05T08:41:26Z INFO  modbus_proxy_rs] modbus connection to 192.168.1.4:502 sucessfull
[2023-02-05T08:44:28Z INFO  modbus_proxy_rs] modbus request 192.168.1.4:502: 12 bytes
[2023-02-05T08:44:28Z DEBUG modbus_proxy_rs] modbus request 192.168.1.4:502: [0, 38, 0, 0, 0, 6, 1, 3, 144, 249, 0, 2]
[2023-02-05T08:44:28Z ERROR modbus_proxy_rs] Client error: RecvError(())
[2023-02-05T08:44:28Z INFO  modbus_proxy_rs] modbus connection to 192.168.1.4:502 sucessfull
```

Any idea what I could do to solve this?

nicx, 2023-02-07:
anoyone here who could help? :)

Gert-dev, 2023-02-10:
I only just started using this project, but I have a SUN-2000L working with modbus-proxy and I have the following settings applied as the defaults didn't work for me:

```yaml
timeout: 3
connection_time: 1
```

This has been working stably for about a week now and works when I use qModMaster to query data.

giss69, 2023-02-15:
@Gert-dev 
Hi, could you please show your config.yml? Where is your container running?
I'm trying to get it work with my SUN2000 10KTL inverter (IP address is 192.168.3.5) on my Synology Docker( 192.168.1.5)
host port 5200 mapped to container port 502.
docker run -d -p 5020:502 -v /volume1/docker/modbus/config.yml:/etc/modbus-proxy.yml tiagocoutinho/modbus-proxy
my config.yml
`devices:
- modbus:
    url: 192.168.3.5:502
    timeout: 3              
    connection_time: 1    
  listen:
    bind: 0:502`

without success. I get this error:
`     
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Os { code: 13, kind: PermissionDenied, message: "Permission denied" }', /modbus-proxy-rs/src/lib.rs:181:67
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace`

Gert-dev, 2023-02-16:
Sure. My configuration:

```yaml
devices:
    - modbus:
          url: 192.168.1.199:502
          timeout: 3
          connection_time: 1
      listen:
          bind: 0.0.0.0:9000
      unit_id_remapping:
          1: 0
logging:
    version: 1
    formatters:
        standard:
            format: "%(asctime)s %(levelname)8s %(name)s: %(message)s"
    handlers:
        console:
            class: logging.StreamHandler
            formatter: standard
    root:
        handlers: ["console"]
        #level: DEBUG
```

I'm not running modbus-proxy in a container, though, just on a Debian host natively using Python 3.

Also remember that if you want to bind to ports under 1024 on Docker you probably need elevated permissions (also one of the reasons I chose 9000), perhaps that's the problem? I don't know what the `python:3.8-alpine` Dockerfile does, but it might be running as a normal user rather than root.

(Finally, not sure if `0` is expanded to `0.0.0.0` in `bind` automatically during listening, but it might.)

giss69, 2023-02-17:
@Gert-dev 
thank you for your help!
But unfortunately it doesn't work for me.
Does anyone has success with Docker and Sun2000?

giss69, 2023-02-17:
@nicx 
Did you get it to work?

nicx, 2023-02-17:
@giss69 dient work here, no change.

But since i use EVCC anyway i tried it with the proxy integrated there: that works wonderfully. modbus-proxy seems to work differently somehow. Too bad that no maintainer reports here, i would have liked to use the solution. :(

nicx, 2023-02-17:
@giss69 just look into the documentation of EVCC, no special needs for the SUN2000, it just worked as expected ;)

XevDoc, 2023-03-22:
@nicx Exactly the same problem, SUN2000 inverter, HA and EVCC. modbus_proxy isn't working for me as well. Thank you for pointing at https://github.com/evcc-io/evcc/pull/4981, will implement this solution as well.