tiagocoutinho/modbus-proxy #32: PermissionError on Port 502

bitwicht, 2024-01-01:
* ModBus TCP proxy version: [v0.6.7]
* Python version:
* Operating System:

I need to start the server on port 502, because i have an device where i cant change the port.

xx@raspberrypi:~/.local/bin $ ./modbus-proxy -b tcp://0:502 --modbus tcp://192.168.0.123:1502
2024-01-01 15:48:23,252     INFO modbus-proxy: Starting...
Traceback (most recent call last):
  File "/home/pi/.local/bin/./modbus-proxy", line 8, in <module>
    sys.exit(main())
  File "/home/pi/.local/lib/python3.9/site-packages/modbus_proxy.py", line 343, in main
    asyncio.run(run())
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/pi/.local/lib/python3.9/site-packages/modbus_proxy.py", line 338, in run
    await run_bridges(bridges, ready=ready)
  File "/home/pi/.local/lib/python3.9/site-packages/modbus_proxy.py", line 327, in run_bridges
    await start_bridges(bridges)
  File "/home/pi/.local/lib/python3.9/site-packages/modbus_proxy.py", line 320, in start_bridges
    await asyncio.gather(*coros)
  File "/home/pi/.local/lib/python3.9/site-packages/modbus_proxy.py", line 188, in start
    self.server = await asyncio.start_server(
  File "/usr/lib/python3.9/asyncio/streams.py", line 94, in start_server
    return await loop.create_server(factory, host, port, **kwds)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 1494, in create_server
    raise OSError(err.errno, 'error while attempting '
PermissionError: [Errno 13] error while attempting to bind on address ('0.0.0.0', 502): permission denied

Other Question:
Modbus Slave is: 192.168.0.123:1502 
Modbus Master is: 192.168.0.11 and an other one ist 
Modbus Master is: 192.168.0.12

I want collect the Modbus Date vom 192.168.0.123:1502 and give it to both masters.

Can i also collect from 192.168.0.123:1502 and open two ports on the proxy 

modbus-proxy -b tcp://0:502 AND tcp://0:503  --modbus tcp://192.168.0.123:1502 ????

If the Master x.x.x.12 is collect from x.x.x.1502 the other master can not collect from x.x.x.1502

Richigeht, 2024-02-12:
Hi

port 502 is system exclusive - run as sudo if you require system ports.

tiagocoutinho, 2024-09-01:
Agree with @Richigeht .

Prefix your command wit `sudo` if you need to serve modbus-proxy on privileged port(s).

closed 2024-09-01 by tiagocoutinho