tiagocoutinho/modbus-proxy #34: Error installing modbus-proxy on Synology with docker compose file

redi78, 2024-01-10:
* ModBus TCP proxy version: latest
* Python version:
* Operating System: Synology DSM 7.2

### Description

I want to install modbus-proxy on my Synology NAS via Docker compose File.

`version: "3"
services:
    modbus-proxy:
        security_opt:
            - seccomp=unconfined
        container_name: modbus-proxy
        image: tiagocoutinho/modbus-proxy:latest
        ports:
        - "5020:502"
        environment:
            - RUST_BACKTRACE=full
        volumes:
            - /volume2/docker/modbus_proxy/config.yml:/etc/modbus-proxy.yml
        restart: always`

### Error Message:

Container will not start and i get following error message:

`thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Os { code: 13, kind: PermissionDenied, message: "Permission denied" }', /modbus-proxy-rs/src/lib.rs:181:67`

Could someone help please?

Thank you :-)

etofi, 2024-05-01:
I use it on my 218+ with this create statement.

```
docker create \
--name=ModbusTCP_Proxy \
--net=macvlan \
--ip 192.168.178.169 \
--restart unless-stopped \
-e TZ=Europe/Berlin \
-v /volume1/docker/ModbusTCP_Proxy/modbus-proxy.yml:/etc/modbus-proxy.yml \
tiagocoutinho/modbus-proxy:latest
```

giss69, 2024-05-01:
> I use it on my 218+ with this create statement.
> 
> ```
> docker create \
> --name=ModbusTCP_Proxy \
> --net=macvlan \
> --ip 192.168.178.169 \
> --restart unless-stopped \
> -e TZ=Europe/Berlin \
> -v /volume1/docker/ModbusTCP_Proxy/modbus-proxy.yml:/etc/modbus-proxy.yml \
> tiagocoutinho/modbus-proxy:latest
> ```

Can you please share your modbus-proxy.yml? ip 192.168.178.169 which ip is it?
Thank you!

giss69, 2024-05-01:
@redi78 
I've same issue. Did you solve this problem?

giss69, 2024-05-01:
but this build:
https://github.com/Akulatraxas/ha-modbusproxy/
(This addon uses the modbus-proxy of tiagocoutinho:)
is working on Synology but only via HA. Why is original not working?

etofi, 2024-05-01:
> > I use it on my 218+ with this create statement.
> > ```
> > docker create \
> > --name=ModbusTCP_Proxy \
> > --net=macvlan \
> > --ip 192.168.178.169 \
> > --restart unless-stopped \
> > -e TZ=Europe/Berlin \
> > -v /volume1/docker/ModbusTCP_Proxy/modbus-proxy.yml:/etc/modbus-proxy.yml \
> > tiagocoutinho/modbus-proxy:latest
> > ```
> 
> Can you please share your modbus-proxy.yml? ip 192.168.178.169 which ip is it? (container ip?) Thank you!

The IP is the macvlan IP of the modbus proxy container.

```
devices:
- modbus:
    url: 192.168.178.50:502
  listen:
    bind: 0:9000
- modbus:
    url: 192.168.178.106:502
  listen:
    bind: 0:9001
```

giss69, 2024-05-01:
Thank you! now the error has gone! but a don't receive any date from invertor.
devices:
- modbus:
    url: 192.168.30.53:502 (my invertor)
  listen:
    bind: 0:9001
Docker running on 192.168.10.5. I'm trying to get date via modus-proxy container with address : 192.168.10.5:9001
Port maping: host:9001, container:502
what is wrong?

etofi, 2024-05-02:
> Thank you! now the error has gone! but a don't receive any data from invertor. devices:
> 
>     * modbus:
>       url: 192.168.30.53:502 (my invertor)
>       listen:
>       bind: 0:9001
>       Docker running on 192.168.10.5. I'm trying to get data via modus-proxy container with address : 192.168.10.5:9001
>       Port maping: host:9001, container:502
>       what is wrong?

Unfortunately, I don't know.
I can only say that I often had problems accessing services when I used the network mode bridge.
That's why I give all important services their own IP address via macvlan - so there are no problems with access or TCP ports.
Maybe you can try it with that.
```
Range 192.168.178.161 - 192.168.178.190 
docker network create -d macvlan -o parent=eth0 --subnet=192.168.178.0/24 --gateway=192.168.178.1 --ip-range=192.168.178.160/27 --aux-address 'host=192.168.178.191' macvlan
```

giss69, 2024-05-02:
I tried it with macvlan too. The same no data.

tiagocoutinho, 2024-09-01:
Please note that the image `tiagocoutinho/modbus-proxy` corresponds to the project [modbus-proxy-rs](https://github.com/tiagocoutinho/modbus-proxy-rs). This is a version of modbus-proxy written in rust that tries to mimic the functionality of modbus-proxy python but it is a complete rewrite so features and well as bugs are eventually different.

I'd like to apologize for the confusion. I should have named that docker image modbus-proxy-rs.

I've just created [modbus-proxy-py](https://hub.docker.com/repository/docker/tiagocoutinho/modbus-proxy-py)

Please give it a try and let me know.

etofi, 2024-09-01:
> I've just created [modbus-proxy-py](https://hub.docker.com/repository/docker/tiagocoutinho/modbus-proxy-py)
> 
> Please give it a try and let me know.

The new version also works for me, but it requires a small adjustment.
The binding of modbus-proxy.yml goes to /config/modbus-proxy.yml

The old version also worked for me - let's wait and see what [giss69](https://github.com/giss69) says.

giss69, 2024-09-02:
@tiagocoutinho 

finaly it works for me! Thank you very match!

tiagocoutinho, 2024-09-04:
Glad I could help. Good luck with your projects! Closing this one

closed 2024-09-04 by tiagocoutinho