tiagocoutinho/modbus-proxy #41: Missing full license text

dwrobel, 2024-09-03:
Would you please consider adding a full text of the license to the sources?

I'd like to package the `modbus-proxy` in Fedora Linux distribution, however one of the requirement is to have a full copy of the license text. Excerpt from the packaging guidelines [1]:

>Packagers should point out to upstream that by not including a proper full license text,
>they are making it difficult or impossible for anyone to comply with their desired license terms.

[1] https://docs.fedoraproject.org/en-US/packaging-guidelines/LicensingGuidelines/#_license_text

tiagocoutinho, 2024-09-05:
It's very kind of you to make a fedora package.

I'm not sure what is needed... is it a full copy of GPL into each source code file?

They already have a reference to the LICENSE:
```python
# Distributed under the GPLv3 license. See LICENSE for more info.
```
Isn't it enough?

dwrobel, 2024-09-05:
>Isn't it enough?

Unfortunately no, reference is not a full copy of the license text. Please have a look at the lines 647-648 of the license text [1] which states that you need to provide a copy of the license along with the program _(note: the canonical url of the license text is the www.gnu.org site [2])_.

You can grab more ideas from the `hello-world` program which also contains the whole copy of the license text in the COPYING [3] file.

[1] https://git.savannah.gnu.org/cgit/hello.git/tree/COPYING#n647
[2] https://www.gnu.org/licenses/gpl-3.0.txt
[3] https://git.savannah.gnu.org/cgit/hello.git/tree/COPYING