tiagocoutinho/modbus-proxy #45: support for plattform linux/arm64/v8

MaGeHome, 2024-11-07:
The image platform is linux/amd64. For my case i need (linux/arm64/v8)

Is it possible to provide a image for linux/arm64/v8?

geduxas, 2024-11-07:
It's python script it should run everywhere where python is running.

MaGeHome, 2024-11-09:
I would have thought so too. But I get the following warning. And no container will be started.

Home@Raspberry:~ $ docker pull tiagocoutinho/modbus-proxy
Using default tag: latest
latest: Pulling from tiagocoutinho/modbus-proxy
Digest: sha256:7bed465df0ec9a445ab7fa2bf66f2c4b436d30e460d6888cfbe7d49556d44bf4
docker.io/tiagocoutinho/modbus-proxy:latest

Home@Raspberry:~ $ docker run -d -p 5020:502 -v $PWD/config.yml:/etc/modbus-proxy.yml tiagocoutinho/modbus-proxy -name modbus-proxy
**WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested**
a5df5161d263b8866e9c103458ae8e1818aa6f69a81284fab6c957cb56fc116c

Home@Raspberry:~ $ docker ps
CONTAINER ID   IMAGE        COMMAND        CREATED       STATUS        PORTS        NAMES

Home@Raspberry:~ $ docker image ls
REPOSITORY                              TAG       IMAGE ID       CREATED       SIZE
tiagocoutinho/modbus-proxy              latest    4c763994912d   2 years ago   2.23MB

Home@Raspberry:~ $ docker pull --platform linux/arm64 tiagocoutinho/modbus-proxy
Using default tag: latest
latest: Pulling from tiagocoutinho/modbus-proxy
Digest: sha256:7bed465df0ec9a445ab7fa2bf66f2c4b436d30e460d6888cfbe7d49556d44bf4
Status: Image is up to date for tiagocoutinho/modbus-proxy:latest
image with reference docker.io/tiagocoutinho/modbus-proxy:latest was found but does not match the specified platform: wanted linux/arm64, actual: linux/amd64

tiagocoutinho, 2024-11-12:
`modbus-proxy` docker image corresponds to the sibling project which is a rust version this one.

please use `modbus-proxy-py` docker image

MaGeHome, 2024-11-12:
Thanks for your support. I tried, but it's the same result

$ docker pull tiagocoutinho/modbus-proxy-py
Using default tag: latest
latest: Pulling from tiagocoutinho/modbus-proxy-py
c6a83fedfae6: Pull complete
6f14051a7863: Pull complete
e1e82c77ac9c: Pull complete
334024c37334: Pull complete
2240f6c2d99c: Pull complete
0db72f319c3e: Pull complete
55bdeb0ffa04: Pull complete
Digest: sha256:3a16bf9cc34b05457e96129ea42adcecda1f51fb88ca689c2b36c15f7f0ddade
Status: Downloaded newer image for tiagocoutinho/modbus-proxy-py:latest
docker.io/tiagocoutinho/modbus-proxy-py:latest

docker run -p 5020:502 -v ./conf.yml:/config/modbus-proxy.yml tiagocoutinho/modbus-proxy-py
**WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested**
exec /usr/local/bin/modbus-proxy: exec format error

MaGeHome, 2024-11-23:
I have now been able to solve it. 
Here is my solution, in case others need it too.

a) Create a Directory
```
mkdir $PWD/Docker/Modbus-Proxy
```
b) Create a Dockerfile
```
sudo nano $PWD/Docker/Modbus-Proxy/Dockerfile
```
```
FROM python:3.12-alpine

# install dependencies to the local user directory
RUN pip --disable-pip-version-check --no-input --no-cache-dir --timeout 10 install modbus-proxy

ENTRYPOINT ["modbus-proxy"]
CMD ["-c", "/config/modbus-proxy.yml"]
```
c) Build an Image 
```
cd $PWD/Docker/Modbus-Proxy
```
```
sudo docker build -t modbus-proxy .
```
d) Start Container
```
sudo docker run --name Modbus-Proxy --hostname Modbus-Container --restart always --detach modbus-proxy -b tcp://0:502 --modbus tcp://**IP-Inverter**:502 --timeout 10 --modbus-connection-time 0.5
```
[image]

closed 2024-11-23 by MaGeHome